nerves-project/nerves_system_br #17: Is there a OS and version that nerves is known to compile successfully on?

jwthomp, 2015-05-02:
I'm trying to build 'nerves_rpi_defconfig' but getting an error:

"fakeroot: preload library `libfakeroot.so' not found, aborting."

I'm trying to build on Ubuntu 14.04. I'd like to first try building on an OS and version that are known to work successfully.

Thank you for any help!

fhunleth, 2015-05-02:
I'm using Ubuntu 14.04, and I build RPi configurations regularly, so I'm not sure why you're getting that error. Is this a fresh Ubuntu install? 

Given the error, I'd first check that libfakeroot existed, but that doesn't make sense that it would be missing. Does it exist in the buildroot/output/host/lib directory? 

Buildroot has a crazy number of messages that flow by on the screen, and my only other thought is that something else failed and for whatever reason, the build scripts kept going. I'd definitely want to fix that if that's the case.

jwthomp, 2015-05-02:
Thanks for getting back to me. I'll dig into it more and let you know! (It's not a fresh install.)

jwthomp, 2015-05-02:
I am currently doing a new build in a fresh 14.04 install. I'll let you know how that goes. As for the buildroot/output/host/lib directory, I have everything down to host but there is no lib directory.

I do however have a buildroot/output/host/usr/lib directory with libfakeroot.so, .la, and -0.so file.

Is that the correct location for the library to be located?

fhunleth, 2015-05-04:
That's the location when I build it. To be honest, I haven't run into issues with the Buildroot fakeroot support, so this is new to me as well. I'll be curious to find out what the root cause is. Maybe some weird PATH or LD_LIBRARY_PATH issue?

jwthomp, 2015-05-04:
I was just able to get a complete build finished on a fresh 14.04 install. I've put together a Dockerfile based on what I needed to do to get the build complete. I'm going to run that now to make sure it works and make it available if it's useful to anyone else.

I'll likely still poke at my other machine a bit to see if I can figure out the issue.

jwthomp, 2015-05-04:
Ok, I'll get something together for a working Docker environment. It may be a couple days as I'm fairly busy with work at the moment, but I'll let you know once I have it in good shape.

fhunleth, 2015-05-04:
No worries - I'm at a conference in Boston this week, so I probably couldn't do much with it anyway. Thank you so much for the help!

fhunleth, 2016-01-12:
Closing old issue.

closed 2016-01-12 by fhunleth