nerves-project/nerves_system_br #22: Upgrade Beaglebone Black configs to support the universal I/O

fhunleth, 2015-12-02:
Currently, adding connecting almost any new hardware to the BeagleBone Black requires some knowledge of device tree when using Nerves. This is pretty difficult for both newcomers and many experienced Linux developers. The Universal I/O device tree looks like a really good way to reduce this pain. See https://github.com/cdsteinkuehler/beaglebone-universal-io.

Another benefit of this change is that it will allow more people to use `nerves_bbb_defconfig` without modification.

fhunleth, 2016-05-01:
Universal I/O is available now. It would be nice to have an Elixir library that implements `config-pin`, though.

closed 2016-05-01 by fhunleth

mobileoverlord, 2016-05-02:
Was this added to nerves_system_br or as part of the defconfig. Could you point to a commit which references its inclusion. Thanks :)

fhunleth, 2016-05-02:
It turns out that universal I/O device tree overlays had been available for a long time since we use the official Beaglebone Linux kernels. I'm actually not sure when it first worked. When I added this issue, I was unfamiliar with universal I/O and didn't know how the BBB were maintaining it. It turns out that Robert Nelson who maintains the Linux kernel for the BBB copies the .dts files from the universal-io repository above to his kernel tree periodically, so they're magically available for us. I usually sync the BBB kernel whenever I do clean builds on my BBB, so we should be close.

mobileoverlord, 2016-05-02:
sounds great, so we want to reimplement config-pin not just wrap it because the utility is not going to be available

fhunleth, 2016-05-02:
We could do either. The utility is much simpler than it looks on the surface, though. The main complexity is that some directory names in `/sys` are dynamically named. I think an Elixir version would be much more understandable, and I hope that someone would write one. However, if we have to call out to a shell script in the interim, so be it.