nerves-project/nerves_system_br #23: Support for reverting back to known-good firmware

fhunleth, 2015-12-02:
The `fwup` configurations for firmware update with Nerves make it possible to revert to previous working firmware, but there's no build-in support for actually doing that. This functionality is needed for the CI framework where it is anticipated that bad firmware images will be loaded on a semi-regular basis. Beyond that, the functionality is required in most production environments especially where manual intervention is impractical.

The upcoming work by @ghitchens on `firmware.ex` will address this.

mobileoverlord, 2016-07-19:
@ghitchens I'll assist in the docs and merging. I would like this to be the start of the `nerves_runtime` which can be included with the `nerves` app containing the build tools.

fhunleth, 2017-12-20:
I have firmware revert support working in most of the officially supported systems and @mobileoverlord added a revert helper function to `Nerves.Runtime`. It's getting close!

fhunleth, 2018-05-01:
`Nerves.Runtime.revert` should get most people there and custom systems can add automatic revert support.

closed 2018-05-01 by fhunleth