nerves-project/nerves_system_br #25: Modify `erlinit.config` on the application side

fhunleth, 2015-12-02:
Currently it's not easy to modify `/etc/erlinit.config` when building an application. This would be useful to enable/disable the hang-on-error feature (e.g. when building dev or prod releases) and also to support non-default tty's, helper programs, and other debugging features.

Fixing this currently requires squashfs overlay merging in Nerves.

kfatehi, 2015-12-27:
At this point, what's the best way to disable hang-on-error?

mobileoverlord, 2016-07-19:
erlinit.config can be overwritten by supplying your own rootfs-additions in your mix project. Documentation for it can be found here: https://hexdocs.pm/nerves/advanced-configuration.html#overwriting-files-in-the-root-file-system

closed 2016-07-19 by mobileoverlord