nerves-project/nerves_system_br #39: Intel Edison port

fhunleth, 2016-01-12:
Support the Intel Edison platform. See http://www.intel.com/content/www/us/en/do-it-yourself/edison.html?wapkw=edison

bbhoss, 2016-01-27:
I have been looking at this some, and apparently a vanilla 4.1 kernel boots on the Edison now. Does that help us at all?

fhunleth, 2016-01-30:
A vanilla kernel helps with pushing the port back up to Buildroot which is good long term. If it is missing major functionality, though, then it may not help us with Nerves. I suppose that it depends on what's missing. Good luck with getting it to work. It looked like you're making good progress based on your posts on Slack.

tslater, 2016-02-28:
Can I read about the progress on slack? I'm very interested in this.

fhunleth, 2016-02-28:
I'm not sure if there's been any progress. If there has been, it hasn't been contributed back yet.

Fwiw, it's been a while since I worked on the Edison. Step 1 was to get it working with Buildroot. My progress on that is at https://github.com/fhunleth/bbb-buildroot-fwup/tree/edison/board/edison. I believe that the `fwup-simple.conf` file worked for creating the raw image file even though I didn't document it.

andy-shev, 2017-03-07:
Just in case I would point out that v4.11-rc1 supports most of significant parts of Edison, The progress is described here: https://edison.internet-share.com/wiki/Using_a_vanilla_Linux_kernel_with_Intel_Edison#What_is_working_and_what_doesn.27t

fhunleth, 2017-06-17:
No reported progress and no work is planned on this from the core Nerves team, so closing.

closed 2017-06-17 by fhunleth

andy-shev, 2017-06-17:
Fine by me. Whoever wants to run Edison, vanilla v4.11 is more or less working, v4.12 will have Bluetooth out of the box.

fhunleth, 2017-06-17:
Good to know. Thanks for the update.