nerves-project/nerves_system_br #40: Sourcing `nerves-env.sh` no longer works

GregMefford, 2016-01-18:
It looks like the change in ce0e77ff69d84fe102477ff5cd847ce2c1ed9b44 makes the `nerves-env.sh` script no longer work.

```
greg@Nerves:~/nerves-system-br$ source nerves-env.sh
dirname: invalid option -- 'b'
Try 'dirname --help' for more information.
basename: invalid option -- 'b'
Try 'basename --help' for more information.
-bash: /home/greg/scripts/nerves-env-helper.sh: No such file or directory
Shell environment NOT updated for Nerves!
```

From what I can tell, this is because we're sourcing the script, so `$0` is `-bash`, which `basename` takes as a parameter `-b`, which doesn't exist.

fhunleth, 2016-01-18:
Thanks for letting me know. Oddly enough, this doesn't fail on my systems. However, getting the source directory from a bash script has always been tricky for me. Could you try the following change on your system: a58d147? I removed the use of $0 which I hope will fix this.

GregMefford, 2016-01-18:
yes, that branch seems to work fine for me on Ubuntu 14.04.

fhunleth, 2016-01-18:
Thanks for testing. That change is now in master.

closed 2016-01-18 by fhunleth