nerves-project/nerves_system_br #41: Apple wired keyboard doesn't work with the RPi 2

fhunleth, 2016-02-07:
The keyboard is detected via USB, but it never gets registered with the input driver.

fhunleth, 2016-02-07:
After some investigation, I'm going to defer more work on this until after upgrading the RPi's Linux kernel to a more recent version. Presumably there are other keyboards that won't work, so if you're having trouble and suspect it's a keyboard issue, please log the type of keyboard here. Cheap keyboards universally appear to work fine.

mobileoverlord, 2016-07-19:
I have tested this on `nerves_system_rpi2` v0.6.0 and it is still an issue.

fhunleth, 2016-08-10:
Apple wired keyboard support is not something anyone is too concerned about. If it comes up, we should add an issue to nerves_system_rpi now that the Raspberry Pi code has been factored out of nerves_system_br.

ghitchens, 2016-08-10:
FYI I will mention that I have an apple wired keyboard that _does_ work properly, so this certainly doesn't apply to all apple wired keyboards.   The new apple wired/wireless keyboard also works properly when plugged in with a USB cable.

rosic, 2016-08-21:
FWIT I tried mine (as that's the only USB one I have) on `nerves_system_rpi` v0.6.1 and it's also properly detected, but doesn't type. Worked on Raspbian.
Anyway, no big deal, remote shell was more convenient so far.

pkazmier, 2016-09-25:
I, too, tried mine (only USB keyboard I have) on `nerves_system_rpi3` v0.6.1. It is detected, but doesn't allow me to type. It did work on Raspbian. Screenshot if it helps:

[image]

fhunleth, 2016-09-25:
I think that I found the issue. This needs to be enabled in the Linux configuration:

[image]

I'm not an Apple person, but I think that "Apple Aluminum" refers to the wired USB keyboard. 

Since modifying the Linux configuration for Nerves is a more advanced topic, I'd recommend finding a cheap USB keyboard. The other option is to invest in a USB->serial cable, since Nerves is so much more enjoyable to work with you can attach to devices through a tty console rather than having to switch to a separate monitor/keyboard.

If anyone watching this bug is in a good place to try enabling `CONFIG_HID_APPLE` and checking that it works, I'll gladly take the PR on the `nerves_system_rpi*` repos.

pkazmier, 2016-09-26:
Well, I had no idea what I was doing, but I was able to build a custom nerves build with that kernel CONFIG_HID_APPLE set. My Apple wired keyboard, which didn't work this morning, is now working tonight. Proof below:

[image]

Now I just need to figure out how to make an example like blinky use the new build I just made ...

pkazmier, 2016-09-26:
@fhunleth I created the three pull requests, but I was only able to test on a rpi3 as I don't own the others to actually validate. I also don't own a cheap USB keyboard to validate that they still work, so maybe I should leave this to the pros. :-)

fhunleth, 2016-09-26:
@pkazmier Great job jumping all the way into building custom system images! Your PRs look good and I can't imagine how you could have broken a regular USB keyboard. Thank you so much for finally putting this issue to rest.

closed 2016-09-26 by fhunleth