nerves-project/nerves_system_br #43: Firmware size grew on the BBB since 0.1.0

fhunleth, 2016-02-26:
After not monitoring the firmware size for a while, it grew - quite a bit:

```
-rw-r--r--   1 garth  staff  15749299 Jan 27 15:19 blinky-bbb-0.1.0.fw
-rw-r--r--   1 garth  staff  23135476 Feb 25 15:59 blinky-bbb.fw
-rw-r--r--   1 garth  staff  27607589 Feb 25 15:12 blinky-rpi2.fw
```
I'm sure that something is mistakenly being included now. 

Also see #24 for detecting this earlier.

fhunleth, 2016-03-02:
It turns out that the firmware size actually shrank 2 MB to around 13 MB since blinky-bbb-0.1.0.fw. The growth of the newer .fw files was due to the Erlang/OTP releases accumulating cruft over time. This has been fixed in `bake` by removing the release directory every build.

closed 2016-03-02 by fhunleth

ghitchens, 2016-03-02:
YEA!