nerves-project/nerves_system_br #45: Issue verifying binary

tungd, 2016-03-05:
I'm using Bake 2.5 on Mac OS X, the `scrub-otp-release.sh` script failed but when I comment out the check, the binary appears to be correct and runs find on device.

```
Updating base firmware image with Erlang release...
scrub-otp-release.sh: ERROR: Unexpected executable format for '.../_nerves-tmp/rootfs-additions/srv/erlang/lib/asn1-4.0.1/priv/lib/asn1rt_nif.so'
   Expected: ELF 32-bit LSB executable, ARM, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 3.12.50
        Got: ELF 32-bit LSB shared object, ARM, version 1 (SYSV)
 This file may have been compiled for the host or a different target.
 Make sure that nerves-env.sh has been sourced and rebuild to fix this.
```

Elixir: 1.2.3
Bake: 2.5
Nerves: `{:nerves, "~> 0.2"}`

ghitchens, 2016-03-05:
I am seeing this same behavior.   I believe this is due to something broken in the latest nerves_system_br (0.4.0-rc1).   You can workaround temporarily until a fix comes in by manually editing your Bakefile.lock to specify 0.3.3 (eg `{rpi,[{'nerves/rpi',<<"0.3.3">>}]}`) .

My Bakefile.lock looks like this::

```
{targets,[{bbb,[{'nerves/bbb',<<"0.3.3">>}]},
          {rpi,[{'nerves/rpi',<<"0.3.3">>}]},
          {rpi2,[{'nerves/rpi2',<<"0.3.3">>}]}]}.
```

Alternatively, you could specify version 0.3.3 in your Makefile, but that will avoid updates to future systems altogether.
I you just modify Bakefile.lock, then once nerves_system_br is updated and released and this issue is resolved, you'll be able to do `bake system update` to update your Bakefile.lock 

Either way, to get the new system, you'll have to do...
```
bake system get
```

tungd, 2016-03-05:
Thanks for the info, here's my investigation: The check for executable type is introduced in commit c2578f3. I think the solution would be one of these:

- Fix `executable_type()`, for example:

        executable_type()
        {
            # Run 'file' on $1 and trim out parts
            # that can vary on a platform
            file -b $1 \
                | sed 's/executable/shared object/g' \
                | sed 's/, BuildID[^,]*,/,/g' \
                | sed 's/, dynamically linked[^,]*,/,/g' \
                | sed 's/, for[^,]*,/,/g' \
                | sed 's/,[^,]*stripped//g'
        }

  Given the strings returned by `file` are:

      ELF 32-bit LSB executable, ARM, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 3.12.50, not stripped
      ELF 32-bit LSB shared object, ARM, version 1 (SYSV), dynamically linked, not stripped

- Add `-shared` flag, so the test program is compiled as a library. I'm not sure we would ever ship executable binaries, if that's the case then we can also strip the "executable"/"shared object" in `executable_type()`. (I'm doing this now, not sure if it work for Linux).

fhunleth, 2016-03-05:
Thanks for the analysis. It was really helpful. I made a slightly different fix and committed it to a branch. See 38890308e80b0d4b977e8a00aafb4c71cbf98d77. Could you try it out?

I'm going to update the default configs that run on travis-ci to exercise this case as well, but I won't have those results for another 30-40 minutes.

fhunleth, 2016-03-05:
It looks like the change works for me on Linux, @mobileoverlord on OSX, and Travis-CI, so I'm marking this as fixed. A `0.4.0-rc2` build will be posted shortly.

closed 2016-03-05 by fhunleth

tungd, 2016-03-06:
Thank you very much, `0.4.0-rc2` works perfectly for me.