nerves-project/nerves_system_br #47: Crosscompilation for Rust

fhunleth, 2016-03-08:
Investigate what it would take to add support for crosscompiling Rust NIFs.

fhunleth, 2016-03-08:
Rust isn't supported in Buildroot yet. It's on their todo list, and if/when it gets included, see what it would take to pass support through to Nerves.

heri16, 2016-07-02:
Where can I find more information on Rust NIFs?

mobileoverlord, 2016-07-19:
@fhunleth What is the status on upstream Rust? Has it made it into a release yet, it might be a case of us just needing to spend time syncing up the tag of Buildroot we are on and getting back to @heri16 regarding testing and integration.

fhunleth, 2016-07-19:
It hasn't been merged upstream yet. I'm not sure what the hold up is - it might just not be a high priority for them. I suspect that if anyone were interested, they could add the patches to nerves_system_br. The complicated part will be figuring out what to do with the crosscompilers. Are they part of the Nerves toolchains or are they downloaded via some other mechanism?

fhunleth, 2018-03-16:
Buildroot 2018.02 has rust support. It's in the 1.0.0 rc builds. There's some work to use it in applications, but at least as far as nerves_system_br is concerned, it's available.

closed 2018-03-16 by fhunleth