nerves-project/nerves_system_br #48: Remove need to put vm.args in rel or check for its existance

fhunleth, 2016-03-12:
Currently, you can see this in `nerves-examples` where the user has to include a `vm.args` in the `rel` directory to launch the Elixir shell. It is very easy to manually delete this file when rebuilding. Doing so causes the Erlang shell to be launched instead. This is confusing to users who don't know the magic to start the Elixir shell. Look into whether there's an alternative way with `exrm` to configure this or if there's a way to detect via a build script for the `vm.args` file. I can't think of a case where a Nerves app would not have a `vm.args` file.

mobileoverlord, 2016-07-19:
Half of this is handled now. When using the new project generator, a vm.args file will automatically be created. The user may still delete it at which point we could ask that the default generated vm.args file contain the line necessary to ensure that the IEx prompt is used. We should ask bitwalker if we should support things this way.

bitwalker, 2016-07-19:
I could certainly add this to exrm/distillery's default vm.args file, seems like a good approach to me!

mobileoverlord, 2016-07-19:
The additions are
```elixir
## Start the Elixir shell
-noshell
-user Elixir.IEx.CLI
-extra --no-halt
```

bitwalker, 2016-07-19:
So it turns out I actually can't do this by default, because it breaks `foreground` amongst other things, because it forces a console. Any reason you aren't using the boot script from exrm?

fhunleth, 2016-07-19:
I can speak to the boot script question. We don't use it since we're not running on a general purpose Linux distribution, and it depends on programs that we may not exist. For example, we don't run bash and have a rather limited shell. We have a separate program called erlinit that sets up the Linux userland environment and starts erlexec.

bitwalker, 2016-07-19:
Ah, makes sense. The shell script used by exrm/distillery is actually a portable one, i.e. it will run under `sh`, but nevertheless, it does use grep/awk in places, so I can see the issue there. With distillery you'll be able to pull a vm.args from wherever including other applications/dependencies and add it to the release as the default one. So presumably, Nerves could just tell users the line to drop in their `rel/config.exs` (or if Nerves is templating that out, just put the line in there yourselves), which adds a template overlay and applies the correct default vm.args, e.g.:

```elixir
# in <nerves>/priv/vm.args
-sname <%= release_name %>
-setcookie <%= release_name %>
-noshell
-user Elixir.IEx.CLI
-extra --no-halt

# in rel/config.exs
release :myapp do
  set overlays: [
    {:template, Path.join("#{:code.priv_dir(:nerves)}", "vm.args"), "releases/<%= release_version %>/vm.args"}]
end
```

fhunleth, 2018-03-16:
Closing old issue.

closed 2018-03-16 by fhunleth