nerves-project/nerves_system_br #50: NIF-based Sqlite integration [from #nerves TODO list]

ChuckLutz, 2016-03-19:
(From the #nerves channel TODO list I've been culling):
Date: 22/23 Feb
Participants: ssache, jschneck
Details:
- Keep up with ssache's Sqlite NIF-based integration progress
  - (19 Mar): Intent was to discuss at 3 Mar meetup; follow up separately?

fhunleth, 2016-04-28:
esqlite builds and runs now with the caveat that it only works on with the SMP version of the beam. On a non-SMP beam, you'll get `enif_send: env==NULL on non-SMP VM`.

The Raspberry Pi 2 and 3 targets both build the SMP version of beam, but almost all other targets don't. To get esqlite to run on them, you currently have to update your `nerves_defconfig` to include `BR2_PACKAGE_ERLANG_SMP=y` and rebuild.

The question is how prevalent is this issue? Should we consider running the SMP version of beam on uniprocessors by default?

mobileoverlord, 2016-04-28:
In addition, you need to force SMP to enable by adding to the top of the `rel/vm.args`
```erlang
## Erlang SMP
-smp enable
```

You can also override the number of schedulers with 
`+S <0 - 1024>` where 0 will use the default

mobileoverlord, 2016-07-19:
I believe SMP has been enabled for all platforms so this could be closed.

closed 2016-07-19 by mobileoverlord