nerves-project/nerves_system_br #53: Downloadable SD images [from #nerves TODO list]

ChuckLutz, 2016-03-19:
(From the #nerves channel TODO list I've been culling):
Date: (?? late Feb)
Participants: joe, gregmefford, jschneck
Details:
- Downloadable SD images
  - .fw smaller than .img
  - travis-ci already archives?
  - Users still need fwup for .fw
  - Public Dropbox? Security...
  - webtorrent.io?

fhunleth, 2016-03-19:
I missed this conversation on Slack, but my preference would be to stay with `.fw` files and make `fwup` easier to install/use if that's the issue. Travis-ci posts artefacts at http://nerves-releases.s3.amazonaws.com/list.html. The `.fw` files boot to an Elixir prompt and if you manually bring up the network, you can remsh into them.

GregMefford, 2016-03-25:
Frank,

The idea was to make it easier for a first-time user to try out a
pre-compiled image without having to install anything, so they have an
immediate quick-win.

Now that there are OS packages for fwup, I feel like that's less of a
problem because it should only take a minute to get to that point.

I still think it makes sense to have a .fw file people can download and try
before they figure out how to build their own, though.

GregMefford, 2016-07-01:
I'd like to go ahead and close this issue for the following reasons:

1. I agree with Frank that it makes more sense to stick with `.fw` files (to be used with `fwup` instead of `.iso` files.
2. We now have easy-to-install packages for `fwup` on Linux and expect to have it working on Windows shortly as well.
3. With Mix integration now, it's much easier to build the example projects, so it doesn't make as much sense to have people download and burn a file when they can learn how to build their own firmware file in literally a few minutes.
4. If we decide that we _do_ want to provide these pre-built firmware files, I think the issue should be against the [nerves-examples repository](https://github.com/nerves-project/nerves-examples) since that's where they would most likely be kept as a Release.

closed 2016-07-01 by GregMefford