nerves-project/nerves_system_br #60: Onion Omega support

fhunleth, 2016-03-19:
Add support for the Onion Omega platform: https://onion.io/

Moved here from https://github.com/nerves-project/nerves-examples/issues/3

ramstein74, 2016-03-20:
You should start a kickstarter project...

fhunleth, 2016-03-21:
Fwiw, I have an Onion Omega on my desk. I think that it would be a great Nerves platform.

ramstein74, 2016-03-21:
I also have another onion in my desk and i regret having bought it.

But if nerves come to onion it would be a different story...

brienw, 2016-06-30:
👍

fhunleth, 2016-06-30:
I'm not planning on porting to the Onion any time soon, but I am making progress on a board which is very similar and normally runs OpenWRT. The docs aren't ready yet, but you can see progress at https://github.com/nerves-project/nerves_system_linkit. If someone is interested in doing an Onion port, that repository would likely help. I don't think that it will be an easy port, though, since the Onion has a relatively small NAND Flash.

ramstein74, 2016-07-20:
new onion omega2 arriving with more memory (128Mb)

https://www.kickstarter.com/projects/onion/omega2-5-iot-computer-with-wi-fi-powered-by-linux?ref=nav_search

fhunleth, 2017-06-17:
No reported progress for a long time and no work is planned on this from the core Nerves team, so closing.

closed 2017-06-17 by fhunleth