nerves-project/nerves_system_br #62: C.H.I.P. Support

CaptChrisD, 2016-03-22:
These have started shipping, and more will be shipped soon. I will be receiving my units in June (hopefully).

These are $9 with built in BT and Wifi and seem like a promising platform. 

I think the linux drivers for the Allwinner R8 may not be as well supported which may be an issue. @fhunleth will be able to comment on this in greater depth.

fhunleth, 2016-03-22:
I haven't seen CHIP support on the Buildroot mailing list yet, but I'll keep an eye out. If the CHIP people post their Linux kernel and any bootloader mods they have, then we should be in good shape even if official support isn't included in Buildroot.

CaptChrisD, 2016-03-23:
Some good references:
https://github.com/NextThingCo/CHIP-buildroot
https://github.com/NextThingCo/CHIP-linux
https://github.com/NextThingCo/CHIP-u-boot

heri16, 2016-07-02:
Having been an early C.H.I.P. Kickstarter backer, I have received a unit in February. I couldn't bear to wait any longer for elixir support and just ported my code to go-Lang. Any news on this. 9 dollars with networking is really perfect for nerves target.

damariei, 2017-01-24:
Any updates on support for this hardware? These seem to be pretty popular and nice compact boards which would be awesome for Nerves.

fhunleth, 2017-01-24:
@damariei Someone just sent me two CHIP boards. I've been busy recently, but I looked at the existing software for it and I definitely think that it's possible to get Nerves on it. The first step is to get Buildroot and fwup building an image that has the root filesystem in a squashfs. I have a branch of fwup that is intended to run ubinize and contain the UBI conf file for making an overarching UBI image with U-boot. The idea is to have a UBIFS volume that has the Linux kernel, two SquashFS volumes that have the A/B rootfs images, and a final UBIFS volume that stores application data. If someone knows how to get BareBox running on the CHIP, we could use it instead of U-Boot to get rid of that first UBIFS volume for the Linux kernel. 

Given that the Nerves port to CHIP is not something that someone's actively working on, another option is to forgo Nerves and just install Erlang and Elixir on the board. You'll still be able to use elixir_ale, nerves_uart, and other embedded Elixir packages.

cc/ @michaelkschmidt

fhunleth, 2017-06-17:
No reported progress and no work is planned on this from the core Nerves team, so closing.

closed 2017-06-17 by fhunleth