nerves-project/nerves_system_br #67: scrub-otp-release.sh is too strict

arjan, 2016-05-19:
I compiled an ELF binary on a different system and put it in the priv/ folder to ship it on the nerves image.
When creating the firmware I got this message:

```
Got:
 ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.32

If binary, expecting:
 ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 3.4.110
```

Only the kernel version differs, so the executable most likely works, but it refused to put it in the firmware.
Is this desirable? Why are we forcing binaries to be built on the same host machine as the firmware?
I cane imagine libraries shipping binaries from other vendors out of our control, even closed-source ones.

fhunleth, 2016-05-19:
Our number one support issue has been people unknowingly mixing and matching compilers. That error check does a pretty good job of catching that. In your case, your application may still not work if the C library is different or any shared libraries it links to have different ABIs (we don't include that in the check, but you get the idea). But I agree that it may work just fine. Also, we're not forcing you to compile things on the same machine - you just need to use the same toolchain. Our toolchains are all downloadable from https://github.com/nerves-project/nerves-toolchain/releases.

If you'd like to submit a PR to make it possible to turn the check off, I think that would be a good addition. Given historical support issues, I'd want the default to be enabled, but the error message should tell the user about the option to disable to check. @mobileoverlord might have some ideas about how to best export that check through to `mix.exs`.

arjan, 2016-05-19:
Thanks for the explanation. It's basically a check whether the right toolchain was used, I see now.
In my case I used a different toolchain (as an experiment) but I guess I can live with this check in this case.
I agree it might be handy to turn off the check for the people who know what they're doing, so lets keep the ticket open for now.

arjan, 2016-07-19:
Maybe making a new ticket is better when the issue arises.

closed 2016-07-19 by arjan