nerves-project/nerves_system_br #75: Fix contents of /etc/os-release

fhunleth, 2016-07-21:
The file currently says:
```
$ cat target/etc/os-release 
NAME=Buildroot
VERSION=2016.02-00007-gedc6a51
ID=buildroot
VERSION_ID=2016.02
PRETTY_NAME="Buildroot 2016.02"
```
This should incorporate the nerves_system_br and system release versions.

mobileoverlord, 2016-12-17:
Should we append to this file with a dump of the loaded Nerves packages at assembly time?

fhunleth, 2016-12-17:
I don't think so. This file has a specific use: https://www.freedesktop.org/software/systemd/man/os-release.html. While we're not technically bound to it with Nerves given how many standard Linux packages we don't include, I have come across other software that parses it. Those libraries/programs may get unhappy if we don't follow the spec.

GregMefford, 2017-01-04:
Do you think it should look something like this?

```
NAME=Nerves
VERSION=0.4.2-05996f6
ID=nerves
ID_LIKE=buildroot
VERSION_ID=0.4.2
PRETTY_NAME="Nerves 0.4.2"
HOME_URL="http://nerves-project.org/"
VARIANT="Raspberry Pi 3 Edition"
VARIANT_ID=rpi3
```

fhunleth, 2018-05-01:
Closing due to lack of interest over the past year. I think that firmware metadata largely replaces this for Nerves-based devices. I can't think of any C programs in regularly use that would use this file.

closed 2018-05-01 by fhunleth