nerves-project/nerves_system_br #109: ERROR: rpi-firmware-b51046a2b2bb69771579a549d157205d9982f858.tar.gz has wrong sha256 hash

webhat, 2017-09-18:
While doing `make` I got the following error:
```
ERROR: rpi-firmware-b51046a2b2bb69771579a549d157205d9982f858.tar.gz has wrong sha256 hash: ERROR: expected: 173c55441a78b5ce9cd7739318f71361b7569597e409c0e002c71e0fd6909481 ERROR: got : 2d2775bcfecd92aec06685efb7461258e996cfbfc6ce9b8a791b73883c7fee4f
```

Once I changed the sha256 hash to `2d2775bcfecd92aec06685efb7461258e996cfbfc6ce9b8a791b73883c7fee4f` it stopped giving me errors during `make`.

https://github.com/nerves-project/nerves_system_br/blob/b14b53f2357c6555a26567511a80851c61aaaddd/patches/buildroot/0012-Raspberry-Pi-Zero-W-combined-update.patch#L115

fhunleth, 2017-09-19:
Wow! I just reproduced this. It looks like the compression from the GitHub download changed just slightly. The original file was 118,906,212 bytes long and the new one is 118,905,706. The untarred contents are identical!

Let me update the SHA to the new value and make a `nerves_system_br` release. Thanks for letting me know about this! 

@entone - This looks like the problem you were telling me about.

entone, 2017-09-19:
Yes, that is definitely it.

closed 2017-09-19 by fhunleth

fhunleth, 2017-09-19:
Release made: https://github.com/nerves-project/nerves_system_br/releases/tag/v0.13.8

PR #110 (2017-09-19): Update rpi-userland and rpi-firmware SHAs
These both magically changed even though their tarball's contents stayed
the same. Presumably Github changed their tar.gz compression settings.

This fixes #109.