nerves-project/nerves_system_br #114: `debug_info` builds not being scrubbed properly

ConnorRigby, 2017-09-29:
* NOTE please feel free to close this if im doing something obviously wrong or don't have enough info *

```
Generated farmbot app
|nerves_bootstrap| Building OTP Release...

==> Cleaning all releases..
==> Clean successful!
using rpi3 - dev configuration.
Updating base firmware image with Erlang release...
scrub-otp-release.sh: ERROR: Unexpected executable format for '/home/connor/farmbot/os/os/_nerves-tmp/rootfs-additions/srv/erlang/lib/nerves_uart-0.1.2/priv/nerves_uart'

Got:
 ELF 32-bit LSB executable, ARM, EABI5 version 1, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.4.110, with debug_info

If binary, expecting:
 ELF 32-bit LSB executable, ARM, EABI5 version 1, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.4.110

or, for static binaries:
 ELF 32-bit LSB executable, ARM, EABI5 version 1, statically linked, for GNU/Linux 3.4.110, with debug_info

If shared library, expecting:
 ELF 32-bit LSB shared object, ARM, EABI5 version 1

 This file may have been compiled for the host or a different target.
 Make sure that nerves-env.sh has been sourced and rebuild to fix this.

** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, raw: true}

```

I am not _exactly_ sure how to reproduce this. It only happens on an old codebase so it may just be an old dep thing that has already been solved. if so please close.

fhunleth, 2017-10-04:
Does your old codebase have this line?

https://github.com/nerves-project/nerves_system_br/blob/master/scripts/scrub-otp-release.sh#L68

fhunleth, 2017-10-04:
Closing since I really think that you have a scrub script that predates the line open. Please reopen if the regex isn't working.

closed 2017-10-04 by fhunleth

ConnorRigby, 2017-10-04:
Oh ok. I must not have that. i'm on a fairly old version of `nerves_system_br`. Thanks!