nerves-project/nerves_system_br #138: scrub-otp-release for Go binaries compiled outside of Buildoot

erauer, 2018-03-22:
I'm compiling Go binaries for ARM outside of Buildroot and the "mix firmware" task is failing because of the check for binary compatibility. 

Updating base firmware image with Erlang release...
scrub-otp-release.sh: ERROR: Unexpected executable format for  ....
Got:
 ELF 32-bit LSB executable, ARM, EABI5 version 1, statically linked
....
or, for static binaries:
 ELF 32-bit LSB executable, ARM, EABI5 version 1, statically linked, for GNU/Linux 4.1.39
....

 This file may have been compiled for the host or a different target.
 Make sure that nerves-env.sh has been sourced and rebuild to fix this.

fhunleth, 2018-03-22:
Note to self:

Replace the call to `file` for checking binary compatibility. This never was ideal. Checking the architecture and flags (hard float vs. soft float for ARM) should catch all of the issues that I remember causing pain. Here are lines for extracting the info (thanks to Buildroot for this):

```sh
LC_ALL=C $READELF -h $FILE_TO_CHECK 2>&1 | sed -r -e '/^  Architecture: +(.+)/!d; s//\1/;' | head -1
LC_ALL=C $READELF -h $FILE_TO_CHECK 2>&1 | sed -r -e '/^  Flags: +(.+)/!d; s//\1/;' | head -1
```

fhunleth, 2018-03-25:
@erauer Could you replace `scrub-otp-release.sh` with the version in PR #140? I think that it should fix the issue that you're seeing while still verifying that there aren't target architecture mismatches.

fhunleth, 2018-03-26:
Update: @mobileoverlord used this change to include Rust binaries in Nerves releases. Go and Rust trigger the same issue, so I went ahead and merged the fix.

commit 0aeb557c (2018-03-26): Call readelf directly for executable arch info

We previously used the host system's file command. Depending on the
host, more or less data would be printed out and we'd have to ignore
some of it. This calls the version of readelf that we include in our
toolchains so it should be much more consistent across platforms. It
also verifies less information: only the platform's architecture and any
flags (hard float vs software float/ABI) are checked. This removes the
Linux kernel header check. While the kernel header check is nice, it
currently triggers false errors for Go and Rust-generated binari [... cut]
closed 2018-03-26 by fhunleth