nerves-project/nerves_system_br #142: scrub-target.sh removes files from custom system rootfs_overlay

mattias01, 2018-04-10:
I've created a custom nerves system based on one of the rpi systems. I've then added files to the rootfs_overlay in my custom system. A few of those files (/etc/services and /etc/protocols) are specified in the scrub-target.sh script.
It seems that scrub-target.sh is run after the custom rootfs_overlay is copied because they are not included in the built firmware, but all other files from the custom root are. I've not managed to figure out from where the board/nerves-common/post-build.sh script, which runs the scrub-target.sh script, is run.

According to the comment in rel2fw.sh, this seems to not be the desired behaviour:
```
# Copy over any rootfs additions from the user
# IMPORTANT: This must be the final step before the merge so that the user can
#            override anything.
```

fhunleth, 2018-04-10:
Thanks for posting about this. That code to remove those files is ancient from when I didn't know how to prevent them from being inserted into the root filesystem by Buildroot. You're correct in pointing out the general issue that scrubbing occurs _after_ the rootfs additions instead of _before_. I can get a quick fix for your particular issue, though, since what we're doing with `/etc/services` and `/etc/protocols` is pretty silly.

mattias01, 2018-04-11:
Thanks for the quick response!
I'm not dependent on a fix for this, so no hurry for my sake :)

mattias01, 2019-12-30:
I'm closing this. I'm not dependent on this and unsure if it's still a problem.

closed 2019-12-30 by mattias01