nerves-project/nerves_system_br #166: Can't use QT 5.10 with qtwebengine

fazibear, 2018-06-10:
After upgrade to `nerves_system_br` `1.2` there is no way to use `qtwebengine` with `QT` `5.10`. You have to downgrade to `5.6 (TLS)`. `qtwebengine` does not appear in `menuconfig`.

Found something like this: `Depends on: ... && (BR2_TOOLCHAIN_HAS_GCC_BUG_85862 [=n] || BR2_PACKAGE_QT5_VERSION_5_6 [=n]) && ...` 

Tried to switch `gcc` to `7.X` but with no luck.

fhunleth, 2018-06-10:
That looks like a regression in Buildroot 2018.05. There was a patch posted to the Buildroot mailing list that looked pretty good so I pulled it in as #167.

@mobileoverlord Did you have a chance to update the `kiosk_system_*` projects for the new `nerves_system_br`? I assume that they'll need this.

I'll push out a new `nerves_system_br` release as soon I hear that this is a good fix.

mobileoverlord, 2018-06-10:
I’ll start a build and get back to you.

PR #167 (2018-06-10): qt5webengine: fix broken enable condition
This should fix #166. I have not tested it, though.
closed 2018-06-10 by fhunleth