nerves-project/nerves_system_br #169: magic version warning

fazibear, 2018-06-11:
After upgrading `nerves_system_br` to `1.2` following warning starts to appear: 

```
Updating base firmware image with Erlang release...
file: compiled magic version [530] does not match with shared library magic version [532]
```

closed 2018-07-05 by fhunleth

fazibear, 2018-07-09:
With latest packages:
``` 
  nerves 1.1.0
  nerves_firmware_ssh 0.3.2
  nerves_network 0.3.6
  nerves_network_interface 0.4.4
  nerves_runtime 0.6.1
  nerves_system_br 1.3.2
  nerves_system_linter 0.3.0
  nerves_toolchain_ctng 1.4.0
  nerves_toolchain_x86_64_unknown_linux_gnu 1.0.0
  nerves_uart 1.2.0
  nerves_wpa_supplicant 0.3.2
```

still have a file warnings:

```
==> Cleaning all releases..
==> Release output directory not found! Nothing to do.
Updating base firmware image with Erlang release...
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
file: compiled magic version [530] does not match with shared library magic version [532]
```

fhunleth, 2018-07-09:
@fazibear Could you find the call to `file` that's causing this? I'm searching through the source code and don't see it. I think your Nerves system might still have the old copy, but given how hard it is to search for `file`, I could imagine that I missed a reference.

fazibear, 2018-07-09:
Sorry about that, missed `nerves_system` deps. Thanks for help!

PR #180 (2018-07-05): Remove call to 'file' in scrub script
While I can't reproduce issue #169, the use of `file` to find ELF
binaries is questionable anyway, so this refactors the script to not
call it. I would expect this to fix #169.
commit 292e7471 (2018-07-05): Remove call to 'file' in scrub script

While I can't reproduce issue #169, the use of `file` to find ELF
binaries is questionable anyway, so this refactors the script to not
call it. I would expect this to fix #169.