nerves-project/nerves_system_br #202: Fix shellcheck warnings on all shell scripts

fhunleth, 2018-08-02:
This project has quite a few shell scripts that are missing double quotes around variables and other issues. It would be great if they could all be fixed since it is highly likely that they're causing problems with paths that contain spaces.

Since it looks like there are quite a few warnings, PRs addressing small pieces (like a file or one type of warning) at a time would be appreciated.

pojiro, 2021-04-23:
I'd like to try to fix this, #202. Tag [v1.15.1](https://github.com/nerves-project/nerves_system_br/tree/v1.15.1) has below shell scripts, and current status is below. I will create PR.

|`find nerves_system_br/ -name "*.sh"`|shellcheck|link to PR|
|-|-|-|
|nerves_system_br/tests/smoke-test/verify.sh|- (no warnings)|- (nothing to do)|
|nerves_system_br/tests/smoke-test/rootfs_overlay/srv/erlang/releases/0.1.0/env.sh|-|-|
|nerves_system_br/tests/smoke-test/run_scrubber.sh|-|-|
|nerves_system_br/tests/run_tests.sh|SC2115||
|nerves_system_br/scripts/scrub-target.sh|-|-|
|nerves_system_br/scripts/mksystem.sh|-|-|
|nerves_system_br/scripts/download-buildroot.sh|-|-|
|nerves_system_br/scripts/rel2fw.sh|-|-|
|nerves_system_br/scripts/ci-deploy.sh|-|-|
|nerves_system_br/scripts/scrub-otp-release.sh|-|-|
|nerves_system_br/scripts/buildroot-state.sh|-|-|
|nerves_system_br/scripts/ci-build.sh|-|-|
|nerves_system_br/scripts/nerves-env-helper.sh|-|-|
|nerves_system_br/scripts/mksyslinuxfs.sh|-|-|
|nerves_system_br/nerves-env.sh|muliple SC warnings||
|nerves_system_br/board/nerves-common/post-createfs.sh|SC2086||
|nerves_system_br/board/nerves-common/post-build.sh|SC2086||
|nerves_system_br/create-build.sh|-|-|
|nerves_system_br/support/docker/nerves_system_br/docker-entrypoint.sh|-|-|

pojiro, 2021-04-24:
I'm not the right person to fix nerves-env.sh because the script is  a little bit complicated and I don't understand the script's intention.

commit 3e8971e7 (2022-09-22): Add shellcheck to CI

Fixes #202 and #203
commit 9a06c182 (2022-09-22): Add shellcheck to CI

Fixes #202 and #203
closed 2022-09-22 by fhunleth