nerves-project/nerves_system_br #240: Use `/usr/bin/env bash` instead of `/bin/bash`

the-kenny, 2018-09-13:
### Current behavior

Multiple scripts use hardcoded `/bin/bash` as their shebang line. This fails for systems where `bash` isn't in `/bin` (one notable example is [NixOS](https://nixos.org)).

### Expected behavior

Shell scripts should use `/usr/bin/env bash` as their shebang line, automatically resolving `bash` from `$PATH`.

This likely isn't limited to `nerves_system_br` but may span multiple projects in this organization. I'll hesitate from opening issues / pull requests for this before you people decide if this will get fixed :-)

fhunleth, 2018-09-13:
Thanks for opening this issue. This seems like an important thing to fix and I don't know of any downsides. Please go ahead and send PRs.

fhunleth, 2018-09-17:
@the-kenny I made a pass at the change. I have a feeling like there's more work to do to get this working on NixOS this is a start. I'll try it locally for a while and if I don't notice any issues or hear otherwise from you, I'll merge it.

the-kenny, 2018-09-17:
Thanks for this! If you need any help on NixOS I’m here to help. I’m a bit busy with work right now, but I’ll try to take a look at your changes tomorrow. If I find some more time I can tackle other issues coming up too.

fhunleth, 2018-10-25:
Closing since the change was made in a past release. If there is more NixOS work, please feel free to open up more issues.

closed 2018-10-25 by fhunleth