nerves-project/nerves_system_br #270: Cannot build projects with file paths that contain spaces

ScrimpyCat, 2018-12-04:
### Environment

* Elixir version (`elixir -v`): 1.7.3
* Nerves environment:
```
Pkg:         nerves_system_br
Vsn:         1.5.3
Type:        system_platform
BuildRunner: {nil, []}
```

### Current behavior

When you run `mix firmware` from a project that contains spaces in its path (yes, I like living on the edge), the script (I believe `rel2fw.sh`) will fail. 

```
|nerves_bootstrap| Building OTP Release...

Updating base firmware image with Erlang release...
rootfs_overlay: /path/to/some does not exist!
** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, raw: true}
```

Where the full path would look like: `/path/to/some folder/project/rootfs_overlay`. And `project` is the actual elixir project folder.

If you happen to have another unrelated folder path: `/path/to/some/`. You'll end up with this output:

```
|nerves_bootstrap| Building OTP Release...

Updating base firmware image with Erlang release...
Copying rootfs_overlay: /path/to/some
rootfs_overlay: folder/project/rootfs_overlay does not exist!
** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, raw: true}
```

### Expected behavior

Should be able to handle spaces in the file path.

### Workaround

Remove any spaces from the file path fixes the problem.

GregMefford, 2018-12-04:
Thanks for reporting this!

Probably related to/the same as https://github.com/nerves-project/nerves/issues/302

fhunleth, 2018-12-05:
I'm convinced that as long as we use shell scripts in our build process, we'll be asymptotically close to having this bug fixed. I found a few more strings that weren't quoted, but I think someone who really wants this to work is going to have to spend some time on this.

Closing since https://github.com/nerves-project/nerves/issues/302 came first. However, this is a really nice and to the point bug report, so thanks for that.

closed 2018-12-05 by fhunleth