nerves-project/nerves_system_br #313: Permission errors when using make

moustacheful, 2019-03-20:
### Environment

* Elixir version (`elixir -v`): 1.8
* Nerves environment: (`mix nerves.env --info`)

  Pkg:         nerves_system_br
  Vsn:         1.7.0
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi_mt7601u
  Vsn:         1.7.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_armv6_rpi_linux_gnueabi
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

* Additional information about your host, target hardware or environment that
  may help

Host is running Solus Linux, target hardware is RPI

### Current behavior

I've been trying to create a custom system using the guide provided on the [Nerves docs](https://hexdocs.pm/nerves/systems.html#customizing-your-own-nerves-system)

After executing `mix nerves.system.shell` from my custom system directory and then `make` within the shell, but after a point it errors out because of an apparent permission error:

```
>>> host-lz4 v1.8.3 Installing to host directory
PATH="/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/bin:/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/sbin:/home/user/google-cloud-sdk/bin:/home/user/.asdf/shims:/home/user/.asdf/bin:/home/user/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/bin:/snap/bin" PKG_CONFIG="/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/lib/pkgconfig:/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/share/pkgconfig" PATH="/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/bin:/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/sbin:/home/user/google-cloud-sdk/bin:/home/user/.asdf/shims:/home/user/.asdf/bin:/home/user/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/bin:/snap/bin" PKG_CONFIG="/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/lib/pkgconfig:/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/share/pkgconfig" AR="/usr/bin/ar" AS="/usr/bin/as" LD="/usr/bin/ld" NM="/usr/bin/nm" CC="/usr/bin/gcc" GCC="/usr/bin/gcc" CXX="/usr/bin/g++" CPP="/usr/bin/cpp" OBJCOPY="/usr/bin/objcopy" RANLIB="/usr/bin/ranlib" CPPFLAGS="-I/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/include" CFLAGS="-O2 -I/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/include" CXXFLAGS="-O2 -I/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/include" LDFLAGS="-L/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/lib -Wl,-rpath,/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/lib" INTLTOOL_PERL=/usr/bin/perl /usr/bin/make -j9 PREFIX=/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host install -C /home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/build/host-lz4-v1.8.3
creating pkgconfig
Installing libraries
Installing headers in /home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host/include
lz4 libraries installed
Installing binaries
install: cannot create regular file '/usr/lib64/erlang/erts-10.2.1/bin/lz4': Permission denied
make[3]: *** [Makefile:150: install] Error 1
make[2]: *** [Makefile:98: install] Error 2
make[1]: *** [package/pkg-generic.mk:252: /home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/build/host-lz4-v1.8.3/.stamp_host_installed] Error 2
make: *** [Makefile:23: _all] Error 2
```

I'm not sure how to continue after this.
I left my system repo here https://github.com/moustacheful/nerves_system_rpi_mt7601u

### Expected behavior

System gets built.

GregMefford, 2019-03-20:
I haven't actually tried building this myself yet, but I did a quick compare against the official system to see what you've changed in yours. I'm curious if you've [manually modified the `nerves_defconfig` here](https://github.com/nerves-project/nerves_system_rpi/compare/master...moustacheful:master#diff-1ff9bb7594783d7b778dc58fc0ca892aR40) as opposed to using `make menuconfig` to select the option. I'd expect to see this file have something like:

```
BR2_PACKAGE_LINUX_FIRMWARE_MEDIATEK_MT7601U=y
```

or

```
BR2_PACKAGE_LINUX_FIRMWARE_MEDIATEK_MT7601U=m
```

as opposed to just the package name on its own line like that.

moustacheful, 2019-03-20:
You're right, but to be completely honest I have no idea what happened there, I added that through `make menuconfig` first, then I must've edited that manually, but it makes no sense. Whelp, saved me some future headaches there, thanks @GregMefford !

moustacheful, 2019-03-20:
Tried building after the change you mentioned and... that worked somehow. That was some incredibly opaque error message.

Thanks again!

moustacheful, 2019-03-21:
Started having this issue again, apparently had nothing to do with the previously mentioned change. Any ideas of where I could start looking for issues?

I've updated the system repo, and I'm just doing the following

```
mix deps.get
mix nerves.system.shell
$ make
```

ConnorRigby, 2019-03-21:
@moustacheful can you check your available disk space? Also are you on OSX or Linux host machine?

ConnorRigby, 2019-03-21:
Sorry just saw this: `Host is running Solus Linux`.

ConnorRigby, 2019-03-21:
I'm compiling this system locally on my machine. Will report back if the error is reproduced.

moustacheful, 2019-03-21:
I'm pretty sure I have plenty of space left, but will get back to you as soon as I can check this

moustacheful, 2019-03-21:
It shouldn't be a disk space, I have about 50% left (around 100gb-ish)

Let me know if you manage to compile it. I don't understand how I got this to work yesterday.

ConnorRigby, 2019-03-21:
Your system just finished compiling for me with no changes. 

`install: cannot create regular file '/usr/lib64/erlang/erts-10.2.1/bin/lz4': Permission denied`
I'm not 100% sure but i think this _should_ be writing to 
`/home/user/projects/nerves_system_rpi_mt7601u/.nerves/artifacts/nerves_system_rpi_mt7601u-portable-1.7.0/host//usr/lib64/erlang/erts-10.2.1/bin/lz4'
`
I think this may be a problem with your distro/configuration. To eliminate that possibility, if you have docker installed you can try the Docker buildrunner.  To do so you can set `build_runner: Nerves.Artifact.BuildRunners.Docker` in your mix.exs file above [this](https://github.com/moustacheful/nerves_system_rpi_mt7601u/blob/master/mix.exs#L37) line

moustacheful, 2019-03-21:
This is the resulting output from my machine when using `Nerves.Artifact.BuildRunners.Docker`.

```
⟩ mix nerves.system.shell
stty sane rows 54 cols 71; stty -echo
export PS1=""; export PS2=""
start() {
echo -e "\e[17F\e[0J\e[1;7m
  Preparing Nerves Shell  \e[0m"
echo -e "\e]0;Nerves Shell\a"
export PS1="\e[1;7m Nerves \e[0;1m \W > \e[0m"
export PS2="\e[1;7m Nerves \e[0;1m \W ..\e[0m"
echo Updating build directory.
echo This will take a while if it is the first time...
/nerves/env/platform/create-build.sh /nerves/env/nerves_system_rpi_mt7601u/nerves_defconfig /nerves/build >/dev/null
stty echo
}; start
:gen_event handler Logger.Backends.Console installed in Logger terminating
   ** (exit) an exception was raised:
                                         ** (RuntimeError) no device registered with the name :user
                                    (logger) lib/logger/backends/console.ex:160: Logger.Backends.Console.async_io/2
                                                    (logger) lib/logger/backends/console.ex:145: Logger.Backends.Console.log_event/5
                                                                     (logger) lib/logger/backends/console.ex:49: Logger.Backends.Console.handle_event/2
                 (stdlib) gen_event.erl:577: :gen_event.server_update/4
                                                                              (stdlib) gen_event.erl:559: :gen_event.server_notify/4
                                                                     (stdlib) gen_event.erl:300: :gen_event.handle_msg/6
                                                         (stdlib) proc_lib.erl:249: :proc_lib.init_p_do_apply/3
                                        stty sane rows 54 cols 71; stty -echo
export PS1=""; export PS2=""
start() {

  Preparing Nerves Shell  

Updating build directory.
This will take a while if it is the first time...
make: *** /nerves/env/platform/buildroot: No such file or directory.  Stop.
 Nerves  build > make
make: *** No targets specified and no makefile found.  Stop.
 Nerves  build > 
```

moustacheful, 2019-03-21:
Another update, just tried building this on a different machine, this time with ubuntu 18.04, and I'm getting the exact same error.
Am I missing something maybe? should I have any environment variables set before executing `nerves.system.shell`?

GregMefford, 2019-03-21:
On Ubuntu (not using Docker), did you get the same permission denied error? You shouldn't need to set any special environment variables, but it's possible that variables you _have_ set on Linux could cause the system shell to behave strangely because it is running a sub-shell.

I think if you were to try to Docker build runner on Ubuntu, it should rule that out and maybe we will learn something more.

So sorry that you're having trouble getting this to work! My guess is that it's something weird with a package that is getting pulled into Buildroot for your custom system.

moustacheful, 2019-03-21:
Yep, i'm getting the same consistent errors on both machines and methods.
I also tried building from the base commit (without my changes) after cleaning and re-fetching dependencies, and I'm getting the same issue.

moustacheful, 2019-03-23:
Managed to get the docker build runner to build!
It had to do with a botched symlink, where `deps/nerves_system_br/buildroot` pointed somewhere on the host machine instead, so docker's fs couldn't resolve it.

This is apparently fixed by cleaning the dependencies (mix deps.clean --all) and re fetching them AFTER setting

`build_runner: Nerves.Artifact.BuildRunners.Docker,`

I'm still curious as of why the other method is not working, so I'll keep at it and will report back if I find something.

fhunleth, 2019-04-28:
@GregMefford @moustacheful Where does this issue stand? Can it be closed?

closed 2019-05-02 by fhunleth