nerves-project/nerves_system_br #326: https://github.com/LeToteTeam/kiosk_system_x86_64

fazibear, 2019-06-05:
### Environment

* Elixir version (`elixir -v`): Elixir 1.8.2 (compiled with Erlang/OTP 20)
* Nerves environment: (`mix nerves.env --info`)
* Ubuntu with erlang-solution elixir and erlang packages. Our system is based on https://github.com/LeToteTeam/kiosk_system_x86_64 with following config:

```
BR2_x86_64=y
BR2_CCACHE=y
BR2_OPTIMIZE_3=y
BR2_GLOBAL_PATCH_DIR="${BR2_EXTERNAL_NERVES_PATH}/patches ${NERVES_DEFCONFIG_DIR}/patches"
BR2_TOOLCHAIN_EXTERNAL=y
BR2_TOOLCHAIN_EXTERNAL_DOWNLOAD=y
BR2_TOOLCHAIN_EXTERNAL_URL="https://github.com/nerves-project/toolchains/releases/download/v1.1.0/nerves_toolchain_x86_64_unknown_linux_gnu-linux_x86_64-1.1.0-3078A48.tar.xz"
BR2_TOOLCHAIN_EXTERNAL_CUSTOM_PREFIX="$(ARCH)-unknown-linux-gnu"
BR2_TOOLCHAIN_EXTERNAL_GCC_7=y
BR2_TOOLCHAIN_EXTERNAL_HEADERS_4_1=y
BR2_TOOLCHAIN_EXTERNAL_CUSTOM_GLIBC=y
BR2_TOOLCHAIN_EXTERNAL_CXX=y
BR2_ROOTFS_SKELETON_CUSTOM=y
BR2_ROOTFS_SKELETON_CUSTOM_PATH="${BR2_EXTERNAL_NERVES_PATH}/board/nerves-common/skeleton"
BR2_INIT_NONE=y
BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_EUDEV=y
BR2_ENABLE_LOCALE_WHITELIST="locale-archive"
BR2_GENERATE_LOCALE="en_US.UTF-8"
BR2_ROOTFS_OVERLAY="${BR2_EXTERNAL_NERVES_PATH}/board/nerves-common/rootfs_overlay ${NERVES_DEFCONFIG_DIR}/rootfs_overlay"
BR2_ROOTFS_POST_BUILD_SCRIPT="${BR2_EXTERNAL_NERVES_PATH}/board/nerves-common/post-build.sh ${NERVES_DEFCONFIG_DIR}/post-build.sh ${NERVES_DEFCONFIG_DIR}/checksum.sh"
BR2_ROOTFS_POST_IMAGE_SCRIPT="${NERVES_DEFCONFIG_DIR}/post-createfs.sh"
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_VERSION=y
BR2_LINUX_KERNEL_CUSTOM_VERSION_VALUE="4.20.17"
BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG=y
BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE="${NERVES_DEFCONFIG_DIR}/linux.defconfig"
BR2_LINUX_KERNEL_INSTALL_TARGET=y
BR2_LINUX_KERNEL_NEEDS_HOST_LIBELF=y
BR2_LINUX_KERNEL_EXT_CUSTOMLOGO=y
BR2_PACKAGE_LINUX_FIRMWARE=y
BR2_PACKAGE_LINUX_FIRMWARE_AMDGPU=y
BR2_PACKAGE_LINUX_FIRMWARE_RTL_8169=y
BR2_PACKAGE_BUSYBOX_CONFIG="${NERVES_DEFCONFIG_DIR}/busybox.defconfig"
BR2_PACKAGE_ALSA_UTILS=y
# BR2_PACKAGE_ALSA_UTILS_ALSAMIXER is not set
BR2_PACKAGE_ALSA_UTILS_AMIXER=y
# BR2_PACKAGE_FFMPEG_FFMPEG is not set
BR2_PACKAGE_E2FSPROGS=y
BR2_PACKAGE_DEJAVU=y
BR2_PACKAGE_CUSTOMLOGO_PATH="${NERVES_DEFCONFIG_DIR}/logo_linux_clut224.ppm"
BR2_PACKAGE_MESA3D=y
BR2_PACKAGE_MESA3D_LLVM=y
BR2_PACKAGE_MESA3D_GALLIUM_DRIVER_RADEONSI=y
BR2_PACKAGE_MESA3D_DRI_DRIVER_RADEON=y
BR2_PACKAGE_MESA3D_OPENGL_EGL=y
BR2_PACKAGE_MESA3D_OPENGL_ES=y
BR2_PACKAGE_QT5=y
BR2_PACKAGE_QT5BASE_SQLITE_SYSTEM=y
BR2_PACKAGE_QT5BASE_LINUXFB=y
BR2_PACKAGE_QT5BASE_DEFAULT_QPA="eglfs"
BR2_PACKAGE_QT5BASE_GIF=y
BR2_PACKAGE_QT5BASE_PNG=y
BR2_PACKAGE_QT5MULTIMEDIA=y
BR2_PACKAGE_QT5WEBENGINE=y
BR2_PACKAGE_XORG7=y
BR2_PACKAGE_XLIB_LIBXV=y
BR2_PACKAGE_IOSTAT=y
BR2_PACKAGE_LSHW=y
BR2_PACKAGE_SYSSTAT=y
# BR2_PACKAGE_ALSA_LIB_ALISP is not set
BR2_PACKAGE_CA_CERTIFICATES=y
BR2_PACKAGE_LIBMNL=y
BR2_PACKAGE_LIBEVDEV=y
BR2_PACKAGE_COLLECTD=y
# BR2_PACKAGE_COLLECTD_LOGFILE is not set
# BR2_PACKAGE_COLLECTD_SYSLOG is not set
BR2_PACKAGE_COLLECTD_CONTEXTSWITCH=y
BR2_PACKAGE_COLLECTD_CPU=y
BR2_PACKAGE_COLLECTD_CPUFREQ=y
BR2_PACKAGE_COLLECTD_CPUSLEEP=y
BR2_PACKAGE_COLLECTD_DISK=y
BR2_PACKAGE_COLLECTD_ENTROPY=y
BR2_PACKAGE_COLLECTD_ETHSTAT=y
BR2_PACKAGE_COLLECTD_INTERFACE=y
BR2_PACKAGE_COLLECTD_IRQ=y
BR2_PACKAGE_COLLECTD_LOAD=y
BR2_PACKAGE_COLLECTD_MEMORY=y
BR2_PACKAGE_COLLECTD_PROCESSES=y
BR2_PACKAGE_COLLECTD_SENSORS=y
BR2_PACKAGE_COLLECTD_SERIAL=y
BR2_PACKAGE_COLLECTD_SWAP=y
BR2_PACKAGE_COLLECTD_TCPCONNS=y
BR2_PACKAGE_COLLECTD_THERMAL=y
BR2_PACKAGE_COLLECTD_UPTIME=y
BR2_PACKAGE_COLLECTD_USERS=y
BR2_PACKAGE_COLLECTD_VMEM=y
BR2_PACKAGE_COLLECTD_NETWORK=y
BR2_PACKAGE_SOCAT=y
BR2_PACKAGE_FILE=y
BR2_PACKAGE_HTOP=y
BR2_PACKAGE_TPM2_TOOLS=y
BR2_PACKAGE_UTIL_LINUX_HWCLOCK=y
# BR2_TARGET_ROOTFS_TAR is not set
BR2_TARGET_GRUB2=y
BR2_TARGET_GRUB2_BUILTIN_MODULES="boot linux ext2 squash4 fat part_msdos normal biosdisk loadenv echo true test sleep"
BR2_NERVES_SYSTEM_NAME="our_system"
BR2_PACKAGE_NERVES_CONFIG=y
```

LLVM is needed for the mesa radeon drivers.

After update to `1.8` system won't compile:

```
CMake Error at cmake/modules/AddLLVM.cmake:457 (add_library):
  Cannot find source file:

    /usr/include/ic.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  cmake/modules/AddLLVM.cmake:628 (llvm_add_library)
  lib/WindowsManifest/CMakeLists.txt:8 (add_llvm_library)
```

```
# ls -la /usr/include/ic.h
lrwxrwxrwx 1 root root 34 May 20 22:09 /usr/include/ic.h -> ../lib/erlang/lib/ic-/include/ic.h
```

The file does not exist, the symlink is wrong.

After a little research this is a file from w very old erlang package. Now it not exist. Why it needs files from root filesystem instead of buildroot?

fazibear, 2019-06-05:
The workaround is to copy this file, but this is not a solution.

There is another compilation error with qt5webengine:

```
>>> qt5webengine 5.12.2 Building..........................................** (Mix) Nerves encountered an error while constructing the artifact
[3881/14546] CXX obj/third_party/leveldatabase/leveldatabase/status.o
[3882/14546] CXX obj/third_party/leveldatabase/leveldatabase/builder.o
[3883/14546] CXX obj/third_party/leveldatabase/leveldatabase/log_writer.o
[3884/14546] CXX obj/third_party/leveldatabase/leveldatabase/dbformat.o
[3885/14546] CXX obj/third_party/leveldatabase/leveldatabase/filename.o
[3886/14546] CXX obj/third_party/leveldatabase/leveldatabase/log_reader.o
[3887/14546] CXX obj/third_party/leveldatabase/leveldatabase/db_iter.o
[3888/14546] CXX obj/third_party/leveldatabase/leveldatabase/leveldb_chrome.o
[3889/14546] CXX obj/third_party/leveldatabase/leveldatabase/dumpfile.o
[3890/14546] CXX obj/third_party/leveldatabase/leveldatabase/memtable.o
[3891/14546] CXX obj/third_party/leveldatabase/leveldatabase/write_batch.o
[3892/14546] CXX obj/third_party/leveldatabase/leveldatabase/table_cache.o
[3893/14546] CXX obj/third_party/leveldatabase/leveldatabase/env.o
[3894/14546] CXX obj/third_party/leveldatabase/leveldatabase/version_edit.o
[3895/14546] CXX obj/third_party/leveldatabase/leveldatabase/db_impl.o
[3896/14546] CXX obj/third_party/leveldatabase/leveldatabase/block.o
[3897/14546] CXX obj/third_party/leveldatabase/leveldatabase/repair.o
[3898/14546] CXX obj/third_party/leveldatabase/leveldatabase/iterator.o
[3899/14546] CXX obj/third_party/leveldatabase/leveldatabase/filter_block.o
[3900/14546] CXX obj/third_party/leveldatabase/leveldatabase/block_builder.o
[3901/14546] CXX obj/third_party/leveldatabase/leveldatabase/format.o
[3902/14546] CXX obj/third_party/leveldatabase/leveldatabase/bloom.o
[3903/14546] CXX obj/third_party/leveldatabase/leveldatabase/memenv.o
[3904/14546] ACTION //third_party/libaom:av1_buildflags(/mnt/2tb/nerves/._nerves/rb/artifacts/rzjb_system_dev_beta-portable-1.0.0/build/qt5webengine-5.12.2/src/toolchain:target)
[3905/14546] STAMP obj/third_party/libaom/av1_buildflags.stamp
[3906/14546] CXX obj/third_party/leveldatabase/leveldatabase/merger.o
[3907/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcinit.o
[3908/14546] CXX obj/third_party/leveldatabase/leveldatabase/two_level_iterator.o
[3909/14546] CXX obj/third_party/leveldatabase/leveldatabase/table_builder.o
[3910/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jccoefct.o
[3911/14546] CXX obj/third_party/leveldatabase/leveldatabase/table.o
[3912/14546] CXX obj/third_party/leveldatabase/leveldatabase/arena.o
[3913/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcapimin.o
[3914/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcapistd.o
[3915/14546] CXX obj/third_party/leveldatabase/leveldatabase/cache.o
[3916/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcdctmgr.o
[3917/14546] CXX obj/third_party/leveldatabase/leveldatabase/version_set.o
[3918/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jddctmgr.o
[3919/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdinput.o
[3920/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdcoefct.o
[3921/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jccolor.o
[3922/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdmainct.o
[3923/14546] CXX obj/third_party/leveldatabase/leveldatabase/coding.o
[3924/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdpostct.o
[3925/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdcolor.o
[3926/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdhuff.o
[3927/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdmaster.o
[3928/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdsample.o
[3929/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdmarker.o
[3930/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdphuff.o
[3931/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jfdctfst.o
[3932/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jfdctflt.o
[3933/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jerror.o
[3934/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jfdctint.o
[3935/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jidctflt.o
[3936/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jidctred.o
[3937/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcmainct.o
[3938/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jidctfst.o
[3939/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcomapi.o
[3940/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jdmerge.o
[3941/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcmarker.o
[3942/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcmaster.o
[3943/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcphuff.o
[3944/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jcparam.o
[... 90 lines cut ...]
                 from ../../3rdparty/chromium/third_party/skia/include/utils/SkCanvasStateUtils.h:11,
                 from ../../3rdparty/chromium/third_party/skia/src/utils/SkCanvasStateUtils.cpp:8:
/mnt/2tb/nerves/._nerves/rb/artifacts/rzjb_system_dev_beta-portable-1.0.0/host/opt/ext-toolchain/x86_64-unknown-linux-gnu/include/c++/7.3.0/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
[3969/14546] CC obj/third_party/libjpeg_turbo/libjpeg/jchuff.o
[3970/14546] CXX obj/components/viz/common/resource_format/resource_format_utils.o
[3971/14546] CXX obj/components/services/font/public/cpp/cpp/mapped_font_file.o
[... cut ...]
make[3]: *** [sub-core-make_first] Error 2
Makefile:48: recipe for target 'sub-src-make_first' failed
make[2]: *** [sub-src-make_first] Error 2
package/pkg-generic.mk:238: recipe for target '/mnt/2tb/nerves/._nerves/rb/artifacts/rzjb_system_dev_beta-portable-1.0.0/build/qt5webengine-5.12.2/.stamp_built' failed
make[1]: *** [/mnt/2tb/nerves/._nerves/rb/artifacts/rzjb_system_dev_beta-portable-1.0.0/build/qt5webengine-5.12.2/.stamp_built] Error 2
Makefile:23: recipe for target '_all' failed
make: *** [_all] Error 2
```

fazibear, 2019-06-05:
Tying to use `nerves_toolchain_x86_64_unknown_linux_gnu 1.2.0`

fazibear, 2019-06-05:
Same error with `1.2.0` toolchain :(

fazibear, 2019-06-05:
Found a similar problem here: https://github.com/buildroot/buildroot/commit/3a0ed5d25e26cb2ec99a1da077c9dcfea67b0ecc

fhunleth, 2019-06-07:
I'm able to reproduce include_next and gcc crashes with the latest Buildroot release. I tried a few things, but I've given up for now. The issues seem to be related to Chromium code like you saw above. I haven't investigated the llvm one.

We decided to make another release of the 1.7.x series to support Erlang/OTP 22 and the upcoming Elixir 1.9 release. It's here: https://github.com/nerves-project/nerves_system_br/releases/tag/v1.7.3. It looks like we'll be sticking with 1.7.x releases for web kiosk projects for the time being.

fhunleth, 2019-06-07:
One more thing to add, there's a configuration option to stay with OTP 21. If you do that, then v1.7.3 should be a very minor update that will pull in security fixes and a minor change to erlinit to support Elixir 1.9 releases (which aren't used yet).

fazibear, 2019-06-08:
Great idea. New buildroot seemed broken. But we really need to try QT 5.12. We freezes `nerves_system_br` at `1.4.2`. This is last version that works for us. Next version does not work. Screen is messed up. Looks like QT version may be broken because this is the only change between those versions. 

Did you pass the qt compilation problem to buildroot tracker or something similar (if they have one) ?

fhunleth, 2019-06-11:
I did not post my results to Buildroot. However, they match https://bugs.busybox.net/show_bug.cgi?id=11776, and that person used non-Nerves toolchains. The next thing that I would have tried is to see if the `raspberrypi3_qt5we_defconfig` builds since that one ships with Buildroot.

Regarding `screen`, can you use `dtach` instead? The other option is to copy/paste the build recipe for the version of `screen` that works for you into your Nerves system, but rename it to something else. Then select it instead of Buildroot's version of screen.

fazibear, 2019-06-11:
Great to know, will follow that bug. 

The `llvm` bug is very simmilar. It's all about `-isystem` parameter.

I'm not meant screen like `screen` application. The problem is described here: https://forum.qt.io/topic/101163/distorted-image-after-upgrade-to-5-11-eglfs-webview

fhunleth, 2019-07-27:
I think we can close this now. My understanding is that webengine is compiling now for x86_64 and rpi3 with Qt 5.13.0.

closed 2019-07-27 by fhunleth