nerves-project/nerves_system_br #349: scrub-otp-release.sh: ERROR: Unexpected executable format [...]

jsimmonds2, 2019-07-29:
### Environment

* Elixir version (`elixir -v`):
* Nerves environment: (`mix nerves.env --info`)
* Additional information about your host, target hardware or environment that
  may help

### Current behavior

Include errors, stacktraces, screenshots or code that may help us reproduce the
issue.

### Expected behavior
**elixir -v**
Erlang/OTP 22 [erts-10.4.3] [source] [64-bit] [smp:2:2] [ds:2:2:10] [async-threads:1] [hipe]

Elixir 1.9.0 (compiled with Erlang/OTP 22)

**mix nerves.env --info**
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.8.5
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi3
  Vsn:         1.8.2
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.2.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.6.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi3
  toolchain:  /home/jhs/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-linux_x86_64-1.2.0
  system:     /home/jhs/.nerves/artifacts/nerves_system_rpi3-portable-1.8.2
  app:        /home/jhs/src/RasPi3BPlus/Blinkers/Nerves/fb2

|nerves_bootstrap| Loadpaths End

Target: Raspberry Pi 3 model B+  . . has until recently run this same project successfully

Current behaviour

[image]

Expected behaviour

Firmware file produced successfully.

fhunleth, 2019-07-29:
It doesn't look like the project that you're building went through the upgrade steps to use `nerves` 1.5. 

Could you confirm that these steps were done? https://hexdocs.pm/nerves/updating-projects.html#updating-from-v1-4-to-v1-5

I'm also going to update the text for option #2 to be more helpful.

jsimmonds2, 2019-07-30:
> Could you confirm that these steps were done? https://hexdocs.pm/nerves/updating-projects.html#updating-from-v1-4-to-v1-5

Confirming that those steps were **incomplete** and **incorrectly implemented**.
My apologies and thanks for your helpful insight.  I'm building again on the latest versions ❤️

closed 2019-07-30 by jsimmonds2