nerves-project/nerves_system_br #408: Build error

fazibear, 2020-01-16:
### Environment

* Elixir version (`elixir -v`): 1.9.4
* Nerves environment: (`mix nerves.env --info`)
* Additional envs:
   - NERVES_DL_DIR: $CI_PROJECT_DIR/.nerves/dl
   - NERVES_ARTIFACTS_DIR: $CI_PROJECT_DIR/.nerves/artifacts
   - NERVES_BR_DL_DIR: $CI_PROJECT_DIR/.nerves/dl

### Current behavior

Trying to build custom nerves system with gitlab-ci. With these environments variables the build fails with the following error: 

```
NERVES_SYSTEM is set to a path which does not exist:
/builds/gamecode/rzjb_system_dev/.nerves/artifacts/rzjb_system_dev-portable-1.0.0
Try running `mix deps.get` to see if this resolves the issue by
downloading the missing artifact.
```

It turns out that there is a cillurar link in artifacts directory:

```
rzjb_system_dev-portable-1.0.0 -> /builds/gamecode/rzjb_system_dev/.nerves/artifacts/rzjb_system_dev-portable-1.0.0
```

### Expected behavior

Don't link if artifact exists

mobileoverlord, 2020-01-16:
There was something strange with the artifact linking that may have surfaced this error. The problem is likely in the artifact resolver in the `:nerves` dependency. I'll do a bit of research to see if I can reproduce it locally. I may move the issue depending on what I discover.

fhunleth, 2022-01-30:
@fazibear I'm closing this since it's now 2 years old. If you remember anything about this that might be useful going forward, please let us know.

closed 2022-01-30 by fhunleth