nerves-project/nerves_system_br #622: Export GCC flags in the toolchain_wrapper for use with NIFs/ports

fhunleth, 2022-01-30:
Everything built by Buildroot is fine. However, some important `CFLAGS` are not exposed when compiling ports and NIFs. These enable features like NEON support and add security improvements.

For example, here are some the flags for the RPi 3 that aren't passed on to NIFs and ports:

```
-mabi=aapcs-linux -mfpu=neon-vfpv4 -marm -fstack-protector-strong -Wl,--build-id=none -ffile-prefix-map=/home/fhunleth/nerves/nerves_system_br/o/rpi3a=buildroot -mfloat-abi=hard -mcpu=cortex-a53 -fPIE -pie -Wl,-z,now -Wl,-z,relro
```

Currently NIFs and ports have to settle for the default `-mfpu=vfpv3-d16`.

ConnorRigby, 2022-01-31:
Where are these stored in Buildroot? i thought i remember you saying they make a little shim executable that sets them. if this is the case, maybe we could run that said executable and store it's output somewhere as well?

fhunleth, 2022-02-05:
They're in the `toolchain-wrapper`. Buildroot symlinks all of the gcc binaries to it and then it calls gcc for real with the arguments forced into the commandline. 

I've been thinking about the fix to this for Nerves and I think we need to extract them and include them in `CFLAGS`, etc. What worried me was that Buildroot knew something we didn't, but I'm coming to the conclusion that they're dealing with such a variety of Makefiles, CMake, etc. that ensuring that `CFLAGS` are always respected is really hard. Our situation with Nerves is different: if `CFLAGS` aren't respected, Erlang ports and NIFS won't compile. Adding additional important flags to `CFLAGS` doesn't seem like it's making anything easier or harder.

fhunleth, 2022-02-10:
Addressed in https://github.com/nerves-project/nerves/pull/698.

closed 2022-02-10 by fhunleth