nerves-project/nerves_system_br #646: /usr/bin/file path is hardcoded somewhere

mayl, 2022-05-24:
### Environment

* Elixir version (`elixir -v`):
```
✦ ❯ elixir -v
Erlang/OTP 24 [erts-12.2] [source] [64-bit] [smp:16:16] [ds:16:16:10] [async-threads:1] [jit]

Elixir 1.13.4 (compiled with Erlang/OTP 24)
```
* Nerves environment: (`mix nerves.env --info`)
```
❯ MIX_TARGET=custom_rpi3 mix nerves.env --info
==> nerves
==> app
|nerves_bootstrap| Environment Package List

  Pkg:         custom_rpi3
  Vsn:         1.18.4
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, [make_args: ["source", "all", "legal-info"]]}

  Pkg:         nerves_system_br
  Vsn:         1.18.6
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_toolchain_armv7_nerves_linux_gnueabihf
  Vsn:         1.5.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.8.5
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   custom_rpi3
  MIX_ENV:      dev

Local Nerves system detected but is not compiled:

  custom_rpi3

|nerves_bootstrap| Environment Variable List
  target:     custom_rpi3
  toolchain:  /home/larry/.nerves/artifacts/nerves_toolchain_armv7_nerves_linux_gnueabihf-linux_x86_64-1.5.0
  system:     /home/larry/.nerves/artifacts/custom_rpi3-portable-1.18.4
  app:        .

|nerves_bootstrap| Loadpaths End
````
* Additional information about your host, target hardware or environment that
  may help

Trying to go through the [custom nerves system tutorial](https://hexdocs.pm/nerves/customizing-systems.html) within a `NixOS` `devshell`.

### Current behavior
Building fails because BR is looking for `/usr/bin/file` instead of `/usr/bin/env file` somewhere (similar to #578).  This seems to be the only issue holding me back from building because if I manually symlink `/usr/bin/file` to the actual path to file the build appears to work (currently still building).  Below is the output from my failed build without the symlink into `/usr/bin/file`

Not sure if this is a Nerves issue or a Buildroot issue, but maybe I can at least get a pointer in the right direction. 

```
❯ MIX_TARGET=custom_rpi3 mix firmware
==> nerves
==> nerves_system_rpi3
Generated nerves_system_rpi3 app
make: Entering directory '/home/larry/projects/testingLiveview/custom_nerves/app/deps/nerves_system_br/buildroot-2021.11.2'
  GEN     /home/larry/projects/testingLiveview/custom_nerves/custom_rpi3/.nerves/artifacts/nerves_system_rpi3-portable-1.18.4/Makefile
#
# configuration written to /home/larry/projects/testingLiveview/custom_nerves/custom_rpi3/.nerves/artifacts/nerves_system_rpi3-portable-1.18.4/.config
#
make: Leaving directory '/home/larry/projects/testingLiveview/custom_nerves/app/deps/nerves_system_br/buildroot-2021.11.2'
------------

Build directory successfully created.

Configuration: /home/larry/projects/testingLiveview/custom_nerves/custom_rpi3/nerves_defconfig

Next, do the following:
   1. cd /home/larry/projects/testingLiveview/custom_nerves/custom_rpi3/.nerves/artifacts/nerves_system_rpi3-portable-1.18.4
   2. make

For additional options, run 'make help' in the build directory.

IMPORTANT: If you update nerves_system_br, you should rerun this script.
could not compile dependency :custom_rpi3, "mix compile" failed. Errors may have been logged above. You can recompile this dependency with "mix deps.compile custom_rpi3", update it with "mix deps.update custom_rpi3" or clean it with "mix deps.clean custom_rpi3"
==> app
** (Mix) Nerves encountered an error while constructing the artifact
"/nix/store/2k0qa3qidn2yn1zrcsffv8zvhncf1qdg-devshell-dir/bin/make -j1 O=/home/larry/projects/testingLiveview/custom_nerves/custom_rpi3/.nerves/artifacts/nerves_system_rpi3-portable-1.18.4 HOSTCC=\"/nix/store/2k0qa3qidn2yn1zrcsffv8zvhncf1qdg-devshell-dir/bin/gcc\" HOSTCXX=\"/nix/store/2k0qa3qidn2yn1zrcsffv8zvhncf1qdg-devshell-dir/bin/g++\" syncconfig\n  GEN     /home/larry/projects/testingLiveview/custom_nerves/custom_rpi3/.nerves/artifacts/nerves_system_rpi3-portable-1.18.4/Makefile\nwhich: no file in (/usr/bin)\n\nYou must install '/usr/bin/file' on your build machine\nmake[1]: *** [support/dependencies/dependencies.mk:27: dependencies] Error 1\nmake: *** [Makefile:23: _all] Error 2\n"
```
Include errors, stacktraces, screenshots or code that may help us reproduce the
issue.

### Expected behavior

Build finds `file` by using `/usr/bin/env file` or just expecting `file` to be in the current `PATH`

fhunleth, 2022-05-24:
Thanks for the bug report. 

I found the reference at https://git.busybox.net/buildroot/tree/support/dependencies/dependencies.sh#n73:

```sh
# 'file' must be present and must be exactly /usr/bin/file,
# otherwise libtool fails in incomprehensible ways.
check_prog_host "/usr/bin/file"
```

The commit is at https://git.busybox.net/buildroot/commit/support/dependencies/dependencies.sh?id=e09e1873a1b5d33fa2e9787b964abb575bc335a4.

That's a 2016 comment, so maybe it's outdated now? If it's not possible for you to put `file` in `/usr/bin/file`, then could you file an issue or send a patch to Buildroot?

mayl, 2022-05-24:
Thanks for digging that up, this does look like an upstream buildroot issue.  We can close this and if I can't get my setup working I'll bring the issue there.

I'm actually not sure if I _need_ a custom image - this is part of a larger effort to get 1-wire working on the Beaglebone.  I wrote up what I've tried so far in a [elixirforum post](https://elixirforum.com/t/beaglebone-1-wire-help/47944) looking for help planning my next steps.  If you have time to take a look I would certainly appreciate it, I'm sure you would have some helpful suggestions.

closed 2022-05-24 by mayl