nerves-project/nerves_system_br #719: Failed too build new poncho project on mac

namxam, 2023-03-10:
### Environment

* Elixir version (`elixir -v`): `Elixir 1.14.3 (compiled with Erlang/OTP 25)`
* Nerves environment: (`mix nerves.env --info`)
```
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_toolchain_aarch64_nerves_linux_gnu
  Vsn:         1.8.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_system_br
  Vsn:         1.22.3
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi4
  Vsn:         1.22.1
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, [make_args: ["source", "all", "legal-info"]]}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.9.3
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

|nerves_bootstrap| Precompile Start

==> nerves_toolchain_aarch64_nerves_linux_gnu
|nerves_bootstrap| Compile.NervesPackage start

|nerves_bootstrap| Compile.NervesPackage end

==> nerves_system_rpi4
|nerves_bootstrap| Compile.NervesPackage start

|nerves_bootstrap| Compile.NervesPackage end

==> firmware
|nerves_bootstrap| Precompile End

Nerves environment
  MIX_TARGET:   rpi4
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi4
  toolchain:  /Users/-----/.nerves/artifacts/nerves_toolchain_aarch64_nerves_linux_gnu-darwin_x86_64-1.8.0
  system:     /Users/-----/.nerves/artifacts/nerves_system_rpi4-portable-1.22.1
  app:        .

|nerves_bootstrap| Loadpaths End
```
* Additional information about your host, target hardware or environment that
  may help

### Current behavior

Building a blank new nerves project works as expected. I can build the firmware. But as soon, as add a phoenix UI as described in https://hexdocs.pm/nerves/user-interfaces.html#phoenix-web-interface, the `mix firmware` task fails with

```
==> nerves
==> firmware

Nerves environment
  MIX_TARGET:   rpi4
  MIX_ENV:      dev

==> ui
Generated ui app
==> firmware
Generated firmware app
|nerves| Building OTP Release...

* skipping runtime configuration (config/runtime.exs not found)
* creating _build/rpi4_dev/rel/firmware/releases/0.1.0/vm.args
Updating base firmware image with Erlang release...
scrub-otp-release.sh: ERROR: Unexpected executable format for '/Users/-----/Projects/iot-commander/firmware/_build/rpi4_dev/_nerves-tmp/rootfs_overlay/srv/erlang/lib/file_system-0.2.10/priv/mac_listener'

Got:
 file:Mach-O 64-bit executable x86_64

Expecting:
 readelf:AArch64;0x0

This file was compiled for the host or a different target and probably
will not work.

Check the following:

1. Are you using a path dependency in your mix deps? If so, run
   'mix clean' in that directory to avoid pulling in any of its
   build products.

2. Did you recently upgrade to Elixir 1.9 or Nerves 1.5?
   Nerves 1.5 adds support for Elixir 1.9 Releases and requires
   you to either add an Elixir 1.9 Release configuration or add
   Distillery as a dependency. Without this, the OTP binaries
   for your build machine will get included incorrectly and cause
   this error. See
   https://hexdocs.pm/nerves/updating-projects.html#updating-from-v1-4-to-v1-5

3. Did you recently upgrade or change your Nerves system? If so,
   try cleaning and rebuilding this project and its deps.

4. Are you building outside of Nerves' mix integration? If so,
   make sure that you've sourced 'nerves-env.sh'.

If you are very sure you know what you are doing, you may place an empty
file in the same directory as the offending file(s) called '.noscrub'.
This will explicitly disable scrubbing for that directory.

If you're still having trouble, please file an issue on Github
at https://github.com/nerves-project/nerves_system_br/issues.

** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, raw: true, line_or_bytes: :line}
```

I tried all steps described and also followed other suggestions, but nothing works. I always end up with the same error. It might be stupid, but I cannot get it to work.

In the phoenix app I have changed `mix.exs` like this:
```
      {:esbuild, "~> 0.5", runtime: Mix.env() == :dev && Mix.target() == :host},
      {:tailwind, "~> 0.1.8", runtime: Mix.env() == :dev && Mix.target() == :host},
```

and in the nerves project I added this:
```
{:ui, path: "../ui", targets: @all_targets, env: Mix.env()},
```

### Expected behavior

I would expect the firmware to be built 😏

jjcarstens, 2023-03-10:
This is a problem with whatever your Phoenix UI configuration is and not with Nerves tooling or poncho organization. Specifically, [you need to disable `code_reloader: false`](https://github.com/nerves-project/nerves_examples/blob/main/hello_phoenix/firmware/config/target.exs#L14-L15). The problem is that the reloader is still being built for your target `rpi4`, but it is using your host tooling to determine how the file listener should be created (i.e. this part of the error `file_system-0.2.10/priv/mac_listener`)

The trouble people usually run into here is that those phoenix configurations need to go in `target.exs`  of the _firmware_ project(or somewhere other config file that gets loaded during firmware compilation). Ensure you have the reloader disabled and try again. Also, you may need to completely clean your `_build/` directory first before trying again

namxam, 2023-03-11:
Oh yes, I disabled tailwind as well, but haven't thought about the code reloader. I guess I will try to submit a PR for the documentation to include an info about that.

closed 2023-03-11 by namxam

namxam, 2023-03-11:
I opened a corresponding pull request at https://github.com/nerves-project/nerves/pull/849