nerves-project/nerves_system_br #741: nerves.system.shell broken on OTP26

CJRChang, 2023-07-12:
### Environment

* Elixir version (`elixir -v`): 1.14
* Nerves environment:
```
  Pkg:         nerves_system_br
  Vsn:         1.23.1
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi4
  Vsn:         1.23.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, [make_args: ["source", "all", "legal-info", "PARALLEL_JOBS=64"]]}

  Pkg:         nerves_toolchain_aarch64_nerves_linux_gnu
  Vsn:         1.8.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.9.3
  Type:        toolchain_platform
  BuildRunner: {nil, []}
```

### Current behavior

When using OTP 26, `mix nerves.system.shell` fails with:
```
==> nerves
sh: line 1: exec: tty_sl: not found
** (ArgumentError) argument error
    :erlang.port_control(#Port<0.8>, 25889124, [])
    (nerves 1.10.2) lib/mix/nerves/shell.ex:106: Mix.Nerves.Shell.get_tty_geometry/1
    (nerves 1.10.2) lib/mix/nerves/shell.ex:49: Mix.Nerves.Shell.open/2
    (nerves_bootstrap 1.11.5) lib/attic/system.shell.ex:74: Mix.Tasks.Nerves.System.Shell.run/1
    (mix 1.14.5) lib/mix/task.ex:421: anonymous fn/3 in Mix.Task.run_task/4
    (mix 1.14.5) lib/mix/cli.ex:84: Mix.CLI.run_task/2
```
`tty_sl` is no longer available, possibly due to the new [OTP26 terminal implementation](https://www.erlang.org/news/164#new-terminal).

### Expected behavior

Successfully enter the nerves system shell.

fhunleth, 2023-07-13:
Hi @CJRChang - thanks for sending the report. We're working our way here from other OTP26 tty changes that impact Nerves.

If you have a Linux machine available, we have an alternate way of building Nerves systems that's described at https://github.com/nerves-project/nerves_systems that may help in the mean time.

CJRChang, 2023-10-03:
Related issue https://github.com/nerves-project/nerves/issues/893
Temporary workaround in https://github.com/nerves-project/nerves/pull/897

fhunleth, 2023-10-03:
Thanks for adding the comments with links to where we're tracking the issue and the workaround, since I missed updating this one. Going to close the issue in favor of tracking in the other location.

closed 2023-10-03 by fhunleth