nerves-project/nerves_system_br #772: rpi3 firmware builds fail on x86_64 macOS

danifv, 2024-01-18:
### Environment

* Elixir version (`elixir -v`):
```
Erlang/OTP 26 [erts-14.2.1] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:1] [jit:ns]

Elixir 1.16.0 (compiled with Erlang/OTP 26)
```
* Nerves environment: (`mix nerves.env --info`)
```
==> nerves
==> blinky

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev
```

### Current behavior

`mix firmware` attempts to use a binary of the wrong architecture.  
The command works as intended in older releases, e.g. 1.24.0. 
Error log:
```|nerves| Building OTP Release...

* [Nerves] validating vm.args
* using config/runtime.exs to configure the release at runtime
* creating _build/rpi3_dev/rel/hello_live_view/releases/0.1.0/vm.args
Updating base firmware image with Erlang release...
scrub-otp-release.sh: ERROR: Unexpected executable format for '/****/_build/rpi3_dev/_nerves-tmp/rootfs_overlay/srv/erlang/lib/file_system-0.2.10/priv/mac_listener'

Got:
 file:Mach-O 64-bit executable x86_64

Expecting:
 readelf:ARM;0x5000400, Version5 EABI, hard-float ABI

This file was compiled for the host or a different target and probably
will not work.
```

### Expected behavior
A new firmware is built.

fhunleth, 2024-01-18:
The `file_system` dependency doesn't support cross-compilation. If this compiled without errors in previous releases, then that was a bug.

Can you see what dependency is bringing in the `file_system` dependency? Perhaps that package can be updated to remove it.

jjcarstens, 2024-01-19:
From the logs, this looks to be `hello_live_view`. `file_system` is typically brought in by LiveView or other Phoenix deps and [is turned off in the example app](https://github.com/nerves-project/nerves_examples/blob/main/hello_live_view/config/runtime.exs#L47).

However, it seems my M1 arm architecture gets away with it being there since the check matches what rpi3 is expecting.

```
> hello_live_view √ % ls hello_live_view.unpacked/rootfs/srv/erlang/lib/file_system-0.2.10/priv/                                                                                                                                     main
inotifywait.exe mac_listener
```

I think the change is to adjust the [`phoenix_live_reload` dep in `nerves_examples/hello_live_view`](https://github.com/nerves-project/nerves_examples/blob/main/hello_live_view/mix.exs#L58) to only be included for the host target