nerves-project/nerves_system_br #781: Firmware builds broken on macOS

th0mas, 2024-02-25:
### Environment

* Elixir version (`elixir -v`):
* Nerves environment: 
```
Nerves environment
MIX_TARGET:   rpi3
MIX_ENV:      dev
```
* Additional information about your host, target hardware or environment that
  may help
  * macOS 13.0, running fish shell

### Current behavior

Running `mix firmware` errors with
```
Updating base firmware image with Erlang release...
Copying rootfs_overlay: /Users/tom/dome_radio/dome_radio_firmware/_build/rpi3_dev/nerves/rootfs_overlay
Copying rootfs_overlay: /Users/tom/dome_radio/dome_radio_firmware/rootfs_overlay
mktemp: illegal option -- p
usage: mktemp [-d] [-q] [-t prefix] [-u] template ...
       mktemp [-d] [-q] [-u] -t prefix
```

### Expected behaviour
* Firmware builds correctly.

It looks like e6135acab77375853b9672462a19eb7c81d870b4 introduced an unsupported `mktemp` flag, breaking builds on macOS

fhunleth, 2024-02-25:
That's odd since that commit was specifically for MacOS users. Could you check where mktemp comes from? I wonder if it's built into a shell and for whatever reason you have an old version?

We could certainly add a fallback for your case since I'm sure you're not alone. Given the history of this line, I'm curious to understand better what's going on.

th0mas, 2024-02-25:
Hmmm - it looks like macOS 14 supports the `-p` flag, it's only versions 13 and below that don't seem to support the `-p` flag.

I'll use this as an excuse to finally update my laptop 😅

guillego, 2024-03-04:
I'm seeing the same error message in macOS 13 when I run mix firmware after bumping my `nerves_system_br` version. I'll also take it as a cue to upgrade to macOS 14 😅.

<details><summary>
This is the manpage for the current mktemp</summary>
<p>
MKTEMP(1)                   General Commands Manual                  MKTEMP(1)

NAME
     mktemp – make temporary file name (unique)

SYNOPSIS
     mktemp [-d] [-q] [-t prefix] [-u] template ...
     mktemp [-d] [-q] [-u] -t prefix

DESCRIPTION
     The mktemp utility takes each of the given file name templates and
     overwrites a portion of it to create a file name.  This file name is
     unique and suitable for use by the application.  The template may be any
     file name with some number of ‘Xs’ appended to it, for example
     /tmp/temp.XXXX.  The trailing ‘Xs’ are replaced with the current process
     number and/or a unique letter combination.  The number of unique file
     names mktemp can return depends on the number of ‘Xs’ provided; six ‘Xs’
     will result in mktemp selecting 1 of 56800235584 (62 ** 6) possible file
     names.  On case-insensitive file systems, the effective number of unique
     names is significantly less; given six ‘Xs’, mktemp will instead select 1
     of 2176782336 (36 ** 6) possible unique file names.

     If mktemp can successfully generate a unique file name, the file is
     created with mode 0600 (unless the -u flag is given) and the filename is
     printed to standard output.

     If the -t prefix option is given, mktemp will generate a template string
     based on the prefix and the _CS_DARWIN_USER_TEMP_DIR configuration
     variable if available.  Fallback locations if _CS_DARWIN_USER_TEMP_DIR is
     not available are TMPDIR and /tmp.  Care should be taken to ensure that
     it is appropriate to use an environment variable potentially supplied by
     the user.

     If no arguments are passed or if only the -d flag is passed mktemp
     behaves as if -t tmp was supplied.

     Any number of temporary files may be created in a single invocation,
     including one based on the internal template resulting from the -t flag.

     The mktemp utility is provided to allow shell scripts to safely use
     temporary files.  Traditionally, many shell scripts take the name of the
     program with the pid as a suffix and use that as a temporary file name.
     This kind of naming scheme is predictable and the race condition it
     creates is easy for an attacker to win.  A safer, though still inferior,
     approach is to make a temporary directory using the same naming scheme.
     While this does allow one to guarantee that a temporary file will not be
     subverted, it still allows a simple denial of service attack.  For these
     reasons it is suggested that mktemp be used instead.

OPTIONS
     The available options are as follows:

     -d      Make a directory instead of a file.

     -q      Fail silently if an error occurs.  This is useful if a script
             does not want error output to go to standard error.

     -t prefix
             Generate a template (using the supplied prefix and TMPDIR if set)
             to create a filename template.

     -u      Operate in “unsafe” mode.  The temp file will be unlinked before
             mktemp exits.  This is slightly better than mktemp(3) but still
             introduces a race condition.  Use of this option is not
             encouraged.

EXIT STATUS
     The mktemp utility exits 0 on success, and >0 if an error occurs.

EXAMPLES
     The following sh(1) fragment illustrates a simple use of mktemp where the
     script should quit if it cannot get a safe temporary file.

           tempfoo=`basename $0`
           TMPFILE=`mktemp /tmp/${tempfoo}.XXXXXX` || exit 1
           echo "program output" >> $TMPFILE

     To allow the use of $TMPDIR:

           tempfoo=`basename $0`
           TMPFILE=`mktemp -t ${tempfoo}` || exit 1
           echo "program output" >> $TMPFILE

     In this case, we want the script to catch the error itself.

           tempfoo=`basename $0`
           TMPFILE=`mktemp -q /tmp/${tempfoo}.XXXXXX`
           if [ $? -ne 0 ]; then
                   echo "$0: Can't create temp file, exiting..."
                   exit 1
           fi

SEE ALSO
     confstr(3), mkdtemp(3), mkstemp(3), mktemp(3), environ(7)

HISTORY
     A mktemp utility appeared in OpenBSD 2.1.  This implementation was
     written independently based on the OpenBSD man page, and first appeared
     in FreeBSD 2.2.7.  This man page is taken from OpenBSD.

macOS 13.0                      August 11, 2014                     macOS 13.0
</p>
</details>

fhunleth, 2024-03-04:
@schrockwell Could you look into making a fix?

guillego, 2024-03-04:
I can confirm that it works well on macOS 14.

It does seem like the previous command worked correctly in macOS 13, what would be the issue with reverting back to it?

fhunleth, 2024-03-04:
No issue in reverting. Lets give @schrockwell or others a couple days to respond to get a fix in. If not, I'll revert the change and publish a new release.

schrockwell, 2024-03-04:
Sorry about this, folks. I installed macOS 13 on a fresh volume, did some testing, and I've come up with a fix in #782.

fhunleth, 2024-03-06:
#782 has been merged.

closed 2024-03-06 by fhunleth