nerves-project/nerves_system_br #813: Error generating firmware for rpi4

OladapoAjala, 2024-08-29:
### Environment

* Elixir version (`elixir -v`):
```
Erlang/OTP 27 [erts-15.0] [source] [64-bit] [smp:16:16] [ds:16:16:10] [async-threads:1] [jit:ns]

Elixir 1.17.2 (compiled with Erlang/OTP 27)
```
* Nerves environment: (`mix nerves.env --info`)
```
Nerves environment
  MIX_TARGET:   rpi4
  MIX_ENV:      dev
```
* Additional information about your host, target hardware or environment that
  may help

### Current behavior

```
creating _build/rpi4_dev/rel/PROJECT_NAME/releases/0.1.0/vm.args
Updating base firmware image with Erlang release...
scrub-otp-release.sh: ERROR: Unexpected executable format for '/home/MY-USER/git/PROJECT_PATH/_build/rpi4_dev/_nerves-tmp/rootfs_overlay/srv/erlang/lib/exla-0.7.3/priv/libexla.so'

Got:
 readelf:Advanced Micro Devices X86-64;0x0

Expecting:
 readelf:AArch64;0x0

This file was compiled for the host or a different target and probably
will not work.

Check the following:

1. Are you using a path dependency in your mix deps? If so, run
   'mix clean' in that directory to avoid pulling in any of its
   build products.

2. Did you recently upgrade to Elixir 1.9 or Nerves 1.5?
   Nerves 1.5 adds support for Elixir 1.9 Releases and requires
   you to either add an Elixir 1.9 Release configuration or add
   Distillery as a dependency. Without this, the OTP binaries
   for your build machine will get included incorrectly and cause
   this error. See
   https://hexdocs.pm/nerves/updating-projects.html#updating-from-v1-4-to-v1-5

3. Did you recently upgrade or change your Nerves system? If so,
   try cleaning and rebuilding this project and its deps.

4. Are you building outside of Nerves' mix integration? If so,
   make sure that you've sourced 'nerves-env.sh'.
```

### Expected behavior

Build the firmware successfully.

fhunleth, 2024-08-29:
The error is due to EXLA. Unfortunately, the [`exla` hex package](https://hex.pm/packages/exla) doesn't support Nerves. The error is saying that the native code was compiled for your x86 machine and not for the ARM-based Raspberry Pi 4.

I have not looked into the EXLA package in a long time so I'm unsure what's involved in making the port these days. Do you know if EXLA works well on Raspberry Pi OS?

OladapoAjala, 2024-08-30:
I haven't used EXLA on Raspberry Pi, I only need to compile my project which uses bumblebee and deploy it to a rpi.

Is there a work around using EXLA?

OladapoAjala, 2024-08-30:
Taking out the EXLA package makes it compile but `Nx.Serving.run(serving, input)` now takes forever. I am guessing because EXLA enables GPU usage.

fhunleth, 2024-08-30:
Yeah, it's going to be really slow without acceleration.

There are a few challenges with EXLA and running it on Nerves and the Pi4. I checked in today on status on the #machine-learning channel on the Erlang Ecosystem slack, and the MLIR and IREE backend work is looking more promising. If you're not on the Slack, it's probably best to join it at https://members.erlef.org/join-us and participate in the development there.

I'm going to close this since there's nothing more that I can do to help you. However, it does look like there's really promising work on the EEF slack in this direction.

closed 2024-08-30 by fhunleth